KiCad/kicad-packages3D #693: LED_SMD.3dshapes/LED_0603_1608Metric double as high as it should

tormodvolden, 2020-05-09:
All SMD LEDs are maybe not the same, but is it a coincidence that the LED model has exactly twice the height (1.10mm) of the LEDs that I am using (0.55mm)?

The LED looks strangely tall when it is rendered in Kicad 5.1.5. I have also imported the .step file into FreeCAD 0.18.4 and measured that the model is really 1.10mm high.

I don't know if many more of the SMD LED models have the same issue. Is there some automated way the height can be extracted (properties of bounding box etc) other than importing each of them into FreeCAD and use the poor measurement tool?

tormodvolden, 2020-05-09:
Hmm... Googling around I see mostly thinner LEDs, but there are some that are actually 1.10mm:
http://www.kingbrightusa.com/product.asp?catalog_name=LED&product_id=AP1608CGCK
https://www.reichelt.com/ch/en/ch/de/smd-led-1608-0603-green-8-mcd-120-smd-led-0603-gn-p31433.html

This is maybe turning into a support issue: How can I get another height for importing the PCB layout into FreeCAD using KiCadStepUp? The latter refuses any scaling factor (i.e. 0.5 on the Z axis).

evanshultz, 2020-05-09:
The footprint name doesn't include the height, so therefore you shouldn't expect it to match any particular part. I expect these are scripted, probably with CadQuery. Searching through the scripts and/or merged PRs you may (possibly...) find answers on why the model is this height.

As for the other questions I do not know.

tormodvolden, 2020-05-09:
I did look at the commit where it was introduced but it was a huge commit with many models and no individual details. Edit: Thanks for the merge request link. There were questions in the commit message and much more information in the merge request, the commit message should have been massaged before they merged it IMO.

I managed to make a new, scaled-down model in FreeCAD using the Draft tool: Clone, set Z scale 0.5 and export the clone.

closed 2020-05-09 by tormodvolden