KiCad/kicad-packages3D #701: Reintegrate 3dmodel for the LED_1W_3W_R8

Franck78, 2020-05-23:
Hello,

this 3d model exists in 4.0 lib:

https://github.com/KiCad/kicad-library/blob/master/modules/packages3d/LEDs.3dshapes/LED_1W_3W_R8.wrl

but is missing in v5 "LED_SMD.3dshapes"

note that the footprint for this LED correctly references that 3D name so that just copying the file where it is expected makes things perfectly displayed.

missing file is LED_1W_3W_R8.step

Franck

evanshultz, 2020-05-26:
@Franck78 
Thanks for noticing this. If you'd make a PR to copy the 3D model it can be reviewed and merged most easily.

Franck78, 2020-05-27:
I would but I have no clue what is the .step file purpose (who is mcad), how to make it, with what tool etc etc.

I thing it takes 2 minutes to do it for someone who knows.

Franck78, 2020-08-16:
may, june,july, still no one to say something (or do) about the stupid single file .step ....

chschlue, 2020-08-16:
Almost four months and still nobody has given in to your demand to create the STEP model for you? What a shame.

Note that we don't take requests. If you want it done, do it yourself. Or pay somebody to do it for you.

Franck78, 2020-08-16:
Now that you spent time writing the useless part of the answer, can you add the useful one ? 

Also, did you notice that the mistake as a "we" mistake ? I don't need that 'fix' to use this model but maybe others do.

chschlue, 2020-08-16:
Did you notice this apparently isn't a mistake, but rather the VRML model didn't make it to v5 precisely *because* the STEP model (and proper source to create one) is missing?

Franck78, 2020-08-16:
No.

So, you commit lot of of models and have no clue how you produce or validate "step". Bad luck for me.

Assuming this is the reason, this Led is  listed somewhere with other problematic components to migrate. If this is the case, it is wiser to fix those components (reintegrate) that may have been use in v4 schematics.

chschlue, 2020-08-16:
Your arrogant ramblings are turning into complete gibberish.
Closing.

closed 2020-08-16 by chschlue

Franck78, 2020-08-16:
The easy way for little dictators: kill, close, ban, erase. For rumblings ? Because I ask the 'step' or the way to build it ??? LOL

Worst, the priorities. Adding new connectors is so much more important than allowing all v4 design to migrate apparently. LOL again.

evanshultz, 2020-08-17:
OK. Everybody, let's take a step back and a deep breath.

@Franck78 
Thanks for taking time to alert us to a missing model for a footprint in the library. This was your first post and you cared enough to come back later and check on things. That's great.

However, we don't take requests. There just isn't enough manpower. The last several months have seen librarian activity drop here, myself including, because of changes in life. I just simply don't have as much time for KiCad these days. Things will return to normal but it's just a slow time around here now.

The 3D models aren't my area of expertise. @chschlue is right that the STEP model is missing at https://github.com/KiCad/kicad-library/tree/master/modules/packages3d/LEDs.3dshapes and that's surely why this model didn't get put into the KiCad v5 library (this repo). https://kicad-pcb.org/libraries/klc/M2.1/ doesn't say that STEP models are required, surprisingly, but in practice that is what I recall seeing. So what we need to get this into the library without any trouble is a good STEP model (I will assume the WRL file you linked to is acceptable).

I believe the script generator for 3D models is at https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/LED_SMD/cq_parameters.yaml. I can see an entry for `LED_1W_3W_R8` but the dimensions are all zero. So I guess the generator is just missing some data for this part or else it would already have been generated. There is no link there for a datasheet or other part information.

The footprint, https://github.com/KiCad/kicad-footprints/blob/master/LED_SMD.pretty/LED_1W_3W_R8.kicad_mod, links to https://www.gme.cz/data/attachments/dsh.518-234.1.pdf. I didn't look, but I assume that's a valid datasheet for this part. I expect you are familiar with this part and would know.

If you'd like to extract the info from the datasheet, add it to the 3D model generator's YAML file, and run the generator script, you should get an acceptable 3D model. I have never done this myself so I don't have much more to tell you about this part of the library, unfortunately. Several contributors and librarians do, though, so surely the readme and asking for a little help, if needed, will be successful if you're willing to put in a little time. You are willing to make a PR and also knowing how to generate 3D models is a useful skill. You are right that there are other people who already know how to do it, but contributions from the community are a large way this library grows and evolves so if you would like to `own` this and fix the issue you pointed out it is a great chance to do just that.

I'll leave this closed since, as stated above, we don't take requests for parts. But we have this record of the missing model if you or anyone else is willing to fill in this gap in the future.

Franck78, 2020-08-17:
Hello Evan,

I took a look at EasyW/kicad-3d-models-in-freecad

The input to this thing is a yaml description of the model (model, sizes, number of pins etc etc etc)
and a python script for several similar models.
The script for this LED model is not written yet.
Probably because it is not a simple rectangular shape with some pins ?

With time I can probably extract python drawing primitives to get the shape. Then Freecad can render the WRML and the STEP.

If there is no specific rules for building a STEP from an existing WRML, I can just use any online converter.

The pdf describes correctly the 1W 3W 5W variant yes perhaps also a 1/2W 2W,  & 4W. But only 1/3/5 commonly available on aliexpress so...

evanshultz, 2020-08-17:
Yes. I'm familiar with the footprint generators and the newer ones also use a YAML file to capture info and a Python generator to make the footprint. Actually, all the footprint and symbol generators are Python it's just that not all footprint generators have a separate YAML file.

I'm not familiar with this part so I don't know anything about it. If the datasheet I linked above is correct it would seem there is some moderate complexity.

I believe it's desirable to use FreeCad as much as possible with the KiCad StepUp plugin. Not required, but nice. And you will need to include the source file in that repo as well. That way, once you've signed the CLA we'll have the source and outputs for the 3D model so someone else will be free to maintain it over time, if needed.

If you have datasheets, which are required for verifying your submission, I don't see why it wouldn't be nice to have all the models you're willing to create. But if there aren't footprints that would also need to be created or else the 3D model is of lesser value. Perhaps just starting with the one missing 3D model is the way to go.

Franck78, 2020-08-18:
(just for latter updates)

https://www.getiangroup.com/data/upload/ueditor/20190909/5d75f9d95ad18.pdf (white)
https://www.getiangroup.com/data/upload/20190902/5d6cc52e92386.pdf  (colors and rgb variant package)

poeschlr, 2020-08-18:
@evanshultz well explained and good notice that the KLC is again written way too polite. I will make a pull request to fix that shortly.

chschlue, 2020-08-18:
@poeschlr don't forget that the whole 3D model section is basically nonsense. See https://github.com/KiCad/kicad-website/issues/435

poeschlr, 2020-08-18:
Damn i had already forgotten about that.

chschlue, 2020-08-18:
@poeschlr BTW, the website is on GL now. Just in case you missed that.

poeschlr, 2020-08-18:
Pull request created here https://gitlab.com/kicad/services/kicad-website/-/merge_requests/502

Franck78, 2020-08-18:
"GL" ? what is this ??

Yes, clarify if there is a need to have the 3d models being obtained by python scripts+yaml from Easy or drawing models with Freecad is enough !

evanshultz, 2020-08-18:
GL = GitLab. It's a different topic. Clarifying the library rules.

The exact tools used aren't going to be specified, but the outputs (STEP required) will be. That's it.

Franck78, 2020-08-27:
(just for latter updates)

https://www.getiangroup.com/data/upload/ueditor/20190909/5d75f9d95ad18.pdf (white)
https://www.getiangroup.com/data/upload/20190902/5d6cc52e92386.pdf (colors and rgb variant package)

another manufacturer with new refs, also for reference
* http://www.kwalityphotonics.com/product-catalogs also known as
* http://www.ledchipindus.com/

with pdf for
* 1w http://www.kwalityphotonics.com/images/pdfdocs/ds_47.1_klsl331we.pdf
* 3w http://www.kwalityphotonics.com/images/pdfdocs/ds_47_klhp3433xe.pdf

Franck78, 2020-08-31:
@evanshultz ,

Hello,

3dmodels for my LED created + updates for existing footprint/symbols.

Also, while at it I created a bunch a scripts cadquerry for some Molex KK-396 series.
* series 47191 footprints OK, need to validate/add the 3dmodels 
* series 47192 same body but Horizontal, new set a footprints needed
* series 5273 a smooth version of 47191, same footprint so just need the new association 

Cadquerry scripts are on Easyw git repo. Who update this git kicad-packages3D repo with new images ?

Where are footprints scripts so that I finish my Molex series ? Who validate and run them ?

Franck

evanshultz, 2020-08-31:
You need to submit the 3D models to this repo. I believe if they're scripted you just need to submit the cadquery stuff. Again, 3D models aren't my expertise in the library.

Footprint scripts are at https://github.com/pointhi/kicad-footprint-generator. Molex connectors may use the SMD single-row script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Connector or a specific script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Connector/Connector_Molex. You may want to extend an existing script since it looks like there are some KK 396 footprints scripts already. Submit any scripts (new or changed) to that repo and submit the footprints at https://github.com/KiCad/kicad-footprints.