KiCad/kicad-packages3D #712: Would like to contribute

MountyRox, 2020-07-30:
Hello KiCad team
I am new to Github and as a first experience I would like to contribute to your KiCad 3D modelling. When looking to the ToDo list there are a lot of items listed, but inspecting the library I found that almost all parts are available already. 

Since I often use KiCad for the layout of dot matrix PCBs and thus I sometimes need a special pitch of several 100 mil. That’s why I started to build my own library and I have built for example standard part models (capacitor, restistor, diode, LED,..) based on a spreadsheet where I easily can change the dimensions of the part and the wire pitch.

Please let me know how I can help.

Best wishes
Joerg

evanshultz, 2020-07-30:
Hi Jeorg,

There are two ways to proceed:
1. Contribute library assets. Simply submit them, broken up into logical divisions, in a PR. No need to ask. Just make a PR.
2. If you have a big plan to discuss first, create an Issue (like you have) with your proposal. Once everyone agrees what to do then you can make a PR.

That's it. You're free to contribute anything as long as you sign the CLA and are willing to make any necessary changes requested so a librarian is ready to merge your PR.

If you have further questions you're free to ask them, but otherwise start making PRs! :) Note that we have only a few librarians, they're all volunteers, and there seems to be little activity lately. So getting your PR reviewed requires patience.

MountyRox, 2020-07-31:
Hi Evan
Thanks for you fast response. A big plan I don’t have.
As soon as I have interesting things to commit, I will just make a PR.

closed 2020-07-31 by MountyRox