KiCad/kicad-packages3D #727: Murata NXE2SxxxxMC has wrong offset and 3D model name

Werni2A, 2020-09-08:
[Converter_DCDC.3dshapes/Converter_DCDC_muRata_NXE2SxxxxMC_THT.wrl](https://github.com/KiCad/kicad-packages3D/blob/master/Converter_DCDC.3dshapes/Converter_DCDC_muRata_NXE2SxxxxMC_THT.wrl) should have an additional offset of around 1.25mm in Y-direction. Furthermore the name of the 3D model is different from the footprint name. When I recall it correctly this is a requirement in the KLC.

Is there a way to find out what the source/generator script of this model is? If there is no commercial software required I will give it a try to fix it :)

[image]

evanshultz, 2020-09-09:
Thanks for reporting this.

The 3D models are not my specialty, but I think many (most?) of them are at https://github.com/easyw/kicad-3d-models-in-freecad. I find an entry for this at https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/Converter_DCDC/cq_parameters.py, but it looks like it's for a THT part yet the pads seem to be SMD. Possibly that's the naming issue you mentioned.

Anyway, I didn't look too deeply into this but hopefully it helps you and points you in the right direction to fix things.

Werni2A, 2020-09-09:
@evanshultz thank you, the script looks right. But how did you find the file? Is there a way in GitHub to search for strings? Or did you clone the repo and searched the local one?

Actually I was talking about the number after NXE, shouldn't it be the same in both parts? I haven't even recognized that one is a THT part and the other one is SMD :see_no_evil: 

Converter_DCDC_muRata_NXE**2**SxxxxMC_THT.wrl
Converter_DCDC_muRata_NXE**x**SxxxxMC_SMD.kicad_mod

I will take a look into this.

evanshultz, 2020-09-09:
I just know there is a lot of the 3D stuff in that repo so I took a look and the Converter_DCDC folder seemed like the one. I just opened the parameter file and text-searched since I was pretty sure that would be it.

I didn't look at the datasheet, and there are probably several things that need fixing. Perhaps the best way, if you're willing to continue, is to fix everything you find and then submit a PR to address it. The fixes are probably straightforward so if you've looking into things and can share what you found was broken and the fix in a PR, it was be verified and merged most easily.

Werni2A, 2020-09-10:
To fix this issue it's probably required to extend the FreeCad [generator script](https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/Converter_DCDC/cq_parameters.py), see [issue](https://github.com/easyw/kicad-3d-models-in-freecad/issues/394).

I will fix the naming inconsistencies and leave the actual issue for someone else to pick up.